jacksonliam/mjpg-streamer #286: mjpg_streamer command enters uninterruptible sleep

DanielYWoo, 2021-01-02:
```
root@cnc:~# mjpg_streamer -i "input_uvc.so -d /dev/video1 -r 640x480 -f 20 -yuv"  -o "output_http.so -p 8082 -w /usr/local/share/mjpg-streamer/www"
MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 20
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
```
the command line hang here, you cannot kill it because it enters uninterruptible sleep status, if you view it by "ps -aux|grep  mjpg", you will see D+
```
root      2644  0.0  0.0      0     0 pts/0    D+   22:32   0:00 [mjpg_streamer]
```

jacksonliam, 2021-02-20:
What hardware/OS/camera is this running on? Can other software (e.g. ffmpeg) capture from the camera successfully?

DanielYWoo, 2021-02-25:
Hardware: Orange Pi PC
Software: 5.4.65-sunxi #2.0.8 SMP (Bionic)
Camera: a cheap USB Camera, no branding.