jacksonliam/mjpg-streamer #287: RG10 image format

ssendev, 2021-01-09:
Hi i have a imx219 "Raspberry" camera but can't get it to work

using no options aborts with `Could not obtain the requested pixelformat: MJPG , driver gave us: RG10` using `-y` or `-u` results in a green image

`v4l2-ctl --list-formats-ext` reports
```
ioctl: VIDIOC_ENUM_FMT
	Index       : 0
	Type        : Video Capture
	Pixel Format: 'RG10'
	Name        : 10-bit Bayer RGRG/GBGB
		Size: Discrete 3264x2464
			Interval: Discrete 0.048s (21.000 fps)
		Size: Discrete 3264x1848
			Interval: Discrete 0.036s (28.000 fps)
		Size: Discrete 1920x1080
			Interval: Discrete 0.033s (30.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
		Size: Discrete 1280x720
			Interval: Discrete 0.017s (60.000 fps)
```

i tried

```diff
diff --git a/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c b/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c
index 4ba1b9d..9a49c0d 100644
--- a/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c
+++ b/mjpg-streamer-experimental/plugins/input_uvc/input_uvc.c
@@ -320,6 +320,8 @@ int input_init(input_parameter *param, int id)
                 format = V4L2_PIX_FMT_RGB24;
             } else if (strcmp(optarg, "RGBP") == 0) {
                 format = V4L2_PIX_FMT_RGB565;
+            } else if (strcmp(optarg, "RG10") == 0) {
+                format = V4L2_PIX_FMT_SRGGB10;
             } else {
               fprintf(stderr," i: FOURCC codec '%s' not supported\n", optarg);
             }
@@ -425,6 +427,9 @@ int input_init(input_parameter *param, int id)
             case V4L2_PIX_FMT_RGB24:
                 fmtString = "RGB24";
                break;
+            case V4L2_PIX_FMT_SRGGB10:
+                fmtString = "RG10";
+                break;
             case V4L2_PIX_FMT_RGB565:
                 fmtString = "RGB565";
                 break;
diff --git a/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c b/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c
index ee99388..ea45cea 100644
--- a/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c
+++ b/mjpg-streamer-experimental/plugins/input_uvc/v4l2uvc.c
@@ -245,6 +245,11 @@ static int init_framebuffer(struct vdIn *vd) {
             vd->framebuffer =
                 (unsigned char *) calloc(1, (size_t) vd->framesizeIn);
             break;
+        case V4L2_PIX_FMT_SRGGB10:
+            vd->framesizeIn = (vd->width * vd->height) * 4*2; // i have no clue but tried diffenrent multipliers: 1,2,3, 3*2, 4*2
+            vd->framebuffer =
+                (unsigned char *) calloc(1, (size_t) vd->framesizeIn);
+            break;
         case V4L2_PIX_FMT_RGB565: // buffer allocation for non varies on frame size formats
         case V4L2_PIX_FMT_YUYV:
         case V4L2_PIX_FMT_UYVY:
```
and then ran it with `LD_LIBRARY_PATH=. ./mjpg_streamer -o "output_http.so -w ./www" -i "input_uvc.so -r 1280x720 -f 60 -fourcc RG10"`

but this results in
```
MJPG Streamer Version: git rev: 5554f42c352ecfa7edaec6fc51e507afce605a34
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: 60
 i: Format............: RG10
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
nvbuf_utils: Could not get EGL display connection
nvbuf_utils: Could not get EGL display connection
Opening in BLOCKING MODE 
 i: The specified resolution is unavailable, using: width 3264 height 2464 instead 
Unable to query that the FPS change is supported
: Inappropriate ioctl for device
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
 i: Error grabbing frames
 i: cleaning up resources allocated by input thread
```

with

```
v4l2-ctl -d /dev/video0 --set-fmt-video=width=3280,height=2464,pixelformat=RG10 --set-ctrl bypass_mode=0 --stream-mmap --stream-count=1 --stream-to=test_3280x2464.bayer
```

i get a file that's 16084992 bytes trying to change the resolution to 720p results in a file of the same size

i also tried to symlink `/opt/vc` as suggested in #259 but that results in a `fatal error: bcm_host.h: No such file or directory` which is caused by me using a Jetson Nano and there appears to be no alternative so the raspicam seems out of the question even on a 32 bit system

jacksonliam, 2021-02-20:
I'm not sure you've implemented everything you need to get RG10 working there. You need to add the bits to convert it to jpeg too. Is that the only pixelformat the camera supports? Can you not get it to give you YUV or RGB565 or something we already support?

ssendev, 2021-02-20:
That was the complete output of `v4l2-ctl --list-formats-ext ` so i would assume it's the only format supported. I tried different format options but that didn't result in an image.

I managed to get something "usable" with https://maker.pro/nvidia-jetson/tutorial/streaming-real-time-video-from-rpi-camera-to-browser-on-jetson-nano-with-flask but i have no idea how that would translate to something that mjpeg-streamer understands.

dongfang-a, 2022-02-11:
Hello,
I have the same issue. 

My background is Arducam announced a "[Sony IMX519](https://www.arducam.com/product/imx519-autofocus-camera-module-for-raspberry-pi-arducam-b0371/)" based PI camera, and they said it is the best camera for raspberry pi and support 10bit image balabala...

Ironically, it does not support either MJPG or YUYV, which made it now nearly useless.

```
v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'pRAA' (10-bit Bayer RGRG/GBGB Packed)
                Size: Discrete 4656x3496
                Size: Discrete 3840x2160
                Size: Discrete 2328x1748
                Size: Discrete 1920x1080
                Size: Discrete 1280x720
        [1]: 'RG10' (10-bit Bayer RGRG/GBGB)
                Size: Discrete 4656x3496
                Size: Discrete 3840x2160
                Size: Discrete 2328x1748
                Size: Discrete 1920x1080
                Size: Discrete 1280x720

```

Kindly could you please check whether you can support this camera?

Here are some other (user) documents about it:
https://www.arducam.com/16mp-autofocus-camera-for-raspberry-pi/
https://www.arducam.com/downloads/arducam-imx519-start-guide.pdf
https://www.arducam.com/docs/cameras-for-raspberry-pi/raspberry-pi-libcamera-guide

jacksonliam, 2022-02-11:
> Hello, I have the same issue.
>
> My background is Arducam announced a "[Sony IMX519](https://www.arducam.com/product/imx519-autofocus-camera-module-for-raspberry-pi-arducam-b0371/)" based PI camera, [...]

Are you using the new Raspbian Buster by any chance? I think you are actually hitting issue #336 not this one. Try enabling the legacy camera stack.

Gusser93, 2022-02-12:
> Are you using the new Raspbian Buster by any chance? I think you are actually hitting issue #336 not this one. Try enabling the legacy camera stack.

I have the same camera and it seems to be not supported by the legacy stack. I tried this on buster and on bullseye with legacy stack enabled. At least for me, the output of v4l2 is the same when enabled and `vcgencmd get_camera` prints `detected=0`.

dongfang-a, 2022-02-13:
> I have the same camera and it seems to be not supported by the legacy stack. I tried this on buster and on bullseye with legacy stack enabled. At least for me, the output of v4l2 is the same when [...]

I tried the same, I have used the Pi Imager to get the latest OS Image both for Raspberry PI OS FULL(32bit) and Raspberry PI OS (LEGACY), install the needed software, but still has the same issue:
`
pi@octoprint:~ $ v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'pRAA' (10-bit Bayer RGRG/GBGB Packed)
                Size: Discrete 4656x3496
                Size: Discrete 3840x2160
                Size: Discrete 2328x1748
                Size: Discrete 1920x1080
                Size: Discrete 1280x720
        [1]: 'RG10' (10-bit Bayer RGRG/GBGB)
                Size: Discrete 4656x3496
                Size: Discrete 3840x2160
                Size: Discrete 2328x1748
                Size: Discrete 1920x1080
                Size: Discrete 1280x720
`