jacksonliam/mjpg-streamer #288: RE: My Command and Issues/Linux SBC

silver2row, 2021-01-20:
Hello,

I have an issue. 

    MJPG Streamer Version: git rev: 501f6362c5afddcfb41055f97ae484252c85c912
     i: Using V4L2 device.: /dev/video0
     i: Desired Resolution: 640 x 480
     i: Frames Per Second.: -1
     i: Format............: YUYV
     i: JPEG Quality......: 80
     i: TV-Norm...........: DEFAULT
     o: www-folder-path......: ./www/
     o: HTTP TCP port........: 8080
     o: HTTP Listen Address..: (null)
     o: username:password....: disabled
     o: commands.............: enabled
     i: select() timeout
     i: cleaning up resources allocated by input thread
    ^Csetting signal to stop
    force cancellation of threads and cleanup resources
     o: cleaning up resources allocated by server thread #00
    done

I am trying to run this command:

    ./mjpg_streamer -i "./input_uvc.so -n -y" -o "./output_http.so -w ./www -p 8080"

Do you see what mistake I am making here? I used to make this source work for running webcams on bots for a Linux SBC that I have currently. 

I picked it up the other day again to test things to make sure I can move on. Oops!

I came up w/ a blank screen and no connection. Please advise me on what I can do to handle this issue. It seems everything is working but the server does not show up.

Seth

P.S. Even when I put the source and webcam in a [image], I get nothing. I am basically receiving a no connection error each time. Has something changed that I am unaware of now or have you moved on from using this excellent source?

jacksonliam, 2021-01-20:
Looks like we are timing out waiting for frames. E.g. the camera is not sending any.

Try adding "-timeout 120" to the input_uvc args?

silver2row, 2021-01-20:
Hello,

Thank you.

Seth

P.S. It turns out that it was my fault (as usual). I had my firewall blocking 8080 for some reason. Sorry sir.

closed 2021-01-20 by silver2row

James4Ever0, 2024-09-08:
@silver2row 

Solution: https://github.com/jacksonliam/mjpg-streamer/issues/182#issuecomment-2336583229