jacksonliam/mjpg-streamer #289: No video stream with input_file.so?

fredrikbaberg, 2021-01-22:
I'm trying to make this work, and I think it has worked in the past, but now I'm encountering an issue.

Everything seems to build correctly. When I launch with: `LD_LIBRARY_PATH=./ ./mjpg_streamer -i "input_file.so -f ./www" -o "output_http.so -w ./www"` I am able to access the page, but can never see the stream or image. The page keeps loading, if I cancel I get a broken image symbol.

I'm using Alpine Linux (for a Docker container) but have also tried it with Ubuntu in WSL2. I'm on Windows 10, using Microsoft Edge as browser but have also tried with Google Chrome.

I'm trying to solve another issue, where the camera module on my Raspberry Pi 3B+ seem to power off after a few seconds, but I wanted to make sure everything with mjpg-streamer works using a file as input first.

Can someone confirm that the stream works for them on a recent install, with input_file.so?

jacksonliam, 2021-01-22:
What are you using to output the jpegs which input_file picks up? Is it definitely writing them to './www' which you have written in the command here?

Seems unusual for ./www to be used as the directory where the jpegs go.

fredrikbaberg, 2021-01-22:
I wanted to keep the example as simple as possible, so that a copy-paste would be enough for testing. That is why I have ./www.

I did not introduce any other files. The folder www is the same as in this repository since I know example.jpg is available there. I can try with another folder with some other jpegs and see if that helps.

jacksonliam, 2021-01-22:
The input file plugin monitors the folder for new jpegs, I believe.

closed 2021-01-22 by jacksonliam