jacksonliam/mjpg-streamer #296: can't compile on raspbian x64

albebert, 2021-03-07:
i'm trying to compile on a x64 rapsbian (pi3) 

first package libjpeg8-dev is not in deb 10 repo so installed libjpeg-dev

but i've error on compilation :  

[ 44%] Linking C shared library input_raspicam.so
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_core.so when searching for -lmmal_core
/usr/bin/ld: cannot find -lmmal_core
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_util.so when searching for -lmmal_util
/usr/bin/ld: cannot find -lmmal_util
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_vc_client.so when searching for -lmmal_vc_client
/usr/bin/ld: cannot find -lmmal_vc_client
/usr/bin/ld: skipping incompatible /opt/vc/lib/libvcos.so when searching for -lvcos
/usr/bin/ld: skipping incompatible /opt/vc/lib/libbcm_host.so when searching for -lbcm_host
collect2: error: ld returned 1 exit status
make[3]: *** [plugins/input_raspicam/CMakeFiles/input_raspicam.dir/build.make:84: plugins/input_raspicam/input_raspicam.so] Error 1
make[3]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[2]: *** [CMakeFiles/Makefile2:256: plugins/input_raspicam/CMakeFiles/input_raspicam.dir/all] Error 2
make[2]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make[1]: *** [Makefile:130: all] Error 2
make[1]: Leaving directory '/home/pi/mjpg-streamer/mjpg-streamer-experimental/_build'
make: *** [Makefile:19: all] Error 2

any idea ?

DiomedesDominguez, 2021-04-16:
Same error

ForsakenRei, 2022-01-20:
This issue seems to be common in different distros as long as they are 64bit, I also saw some other issues states the same problem like this one https://github.com/jacksonliam/mjpg-streamer/issues/259
It's all related to MMAL support on 64bit system so...only thing we can do is wait, unless you want to switch to 32bit.

Skiddz, 2022-02-26:
Having the same issue as the OP, but haven't seen any updates on this issue - this seems to be the most current thread on the subject.  Any news?

Mactarvish, 2022-09-28:
same issue

johnnycash69, 2023-01-10:
Any news? I still can't compile on 64-bit systems.

jacksonliam, 2023-01-10:
I don't believe raspberry pi mmal works on 64 bit so input_raspicam won't work anyway. It's unlikely support will be added since it's now legacy to libcamera (which doesn't support hardware mjpg encoding). 

So just remove `add_subdirectory(plugins/input_raspicam)` from CMakeLists.txt, run a `make distclean` and build again to use input_uvc.

There's probably something we can do with cmake to stop this error on 64 bit systems, one of the PRs that change the files searched might help there.

johnnycash69, 2023-01-10:
> I don't believe raspberry pi mmal works on 64 bit so input_raspicam won't work anyway. It's unlikely support will be added since it's now legacy to libcamera (which doesn't support hardware mjpg [...]

Thanks. Compiling worked and the camera is now up and streaming 👍