jacksonliam/mjpg-streamer #299: Octopi - mjpg-stream cuts off

GandalftheGay, 2021-04-01:
Hello and thanks for the help in advance!

i have Octoprint installed on a raspberry pi 4 and the mjpg-stream come pre-installed with it nobody in there github could help me, so i will try it here, because at this point i think its a problem with the stream and not with the octoprint installation itself.

My Problem:
When i open the IP to watch the stream directly it works perfectly for 10-15 seconds and freezes afterwards.
if i reload the site it works again for 10-15 seconds.

i tried:
- lower framerates and resolution
- changed my wlan-router for a newer one (wanted to upgrade anyway)
- different Browsers (chrome, Firefox and Edge)
- watching the stream via VLC (the stream just closes after 10-15 seconds)

what can i try to solve this problem? do you need any logfiles to help me?
if yes please tell me how i would find them.

Thanks guys!

mbgroen, 2021-05-28:
Same issue here, except reloading the page, unplugging and reconnecting the camera  doesn't work either. After rebooting the raspberry pi (3b+) the camera works again (Microsoft Lifecam Studio HD USB camera). I've  also tried different resolutions. I'm on the latest release of OctoPi and have upgraded all packages to the latest versions.

toufou54, 2021-05-31:
Hi,

I did have sales issues a few months ago..
I did reinstall a full clean Debian distro and octoprint from GitHub (not
the octoprint SD image) and same for mjpeg streamer.

Since then no issues

Sam

mbgroen, 2021-05-31:
Thanks for you’re response. Will give that a try.