jacksonliam/mjpg-streamer #304: Problem with webcam

miko671995, 2021-04-29:
Hello. I have a problem with streaming on my orange pi pc. i use the lastest armbian system. 

there is my first problem - 

`root@orangepipc:/tmp/mjpg-streamer/mjpg-streamer-experimental#  ./mjpg_streamer -i "input_uvc.so  -d /dev/video1" -o "output_http.so"
MJPG Streamer Version: git rev: 310b29f4a94c46652b20c4b7b6e5cf24e532af39
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: Could not obtain the requested pixelformat: MJPG , driver gave us: RGB3
    ... will try to handle this by checking against supported formats.
    ... Falling back to RGB24 mode (consider using -fourcc RGB24 option). Note that this requires much more CPU power
Unable to map buffer: Invalid argument
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
`

after trying adding -fourcc RGB 

`root@orangepipc:/tmp/mjpg-streamer/mjpg-streamer-experimental# ./mjpg_streamer -i "input_uvc.so -fourcc RGB24 -d /dev/video1" -o "output_http.so"
MJPG Streamer Version: git rev: 310b29f4a94c46652b20c4b7b6e5cf24e532af39
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: RGB24
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
Unable to map buffer: Invalid argument
Init v4L2 failed !! exit fatal
 i: init_VideoIn failed
`
`root@orangepipc:/tmp/mjpg-streamer/mjpg-streamer-experimental# lsusb
Bus 008 Device 002: ID 093a:2625 Pixart Imaging, Inc.
`
`For more details see dmesg(1).
root@orangepipc:/tmp/mjpg-streamer/mjpg-streamer-experimental# dmesg
[    0.000000] Booting Linux on physical CPU 0x0
[    0.000000] Linux version 5.10.21-sunxi (root@hirsute) (arm-linux-gnueabihf-gcc (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 8.3.0, GNU ld (GNU Toolchain for the A-profile Architecture 8.3-2019.03 (arm-rel-8.36)) 2.32.0.20190321) #21.02.3 SMP Mon Mar 8 00:28:04 UTC 2021
[    0.000000] CPU: ARMv7 Processor [410fc075] revision 5 (ARMv7), cr=50c5387d
[    0.000000] CPU: div instructions available: patching division code
[    0.000000] CPU: PIPT / VIPT nonaliasing data cache, VIPT aliasing instruction cache
[    0.000000] OF: fdt: Machine model: Xunlong Orange Pi PC
[    0.000000] Memory policy: Data cache writealloc
[    0.000000] cma: Reserved 128 MiB at 0x77c00000
[    0.000000] Zone ranges:
[    0.000000]   Normal   [mem 0x0000000040000000-0x000000006fffffff]
[    0.000000]   HighMem  [mem 0x0000000070000000-0x000000007fffffff]
[    0.000000] Movable zone start for each node
[    0.000000] Early memory node ranges
[    0.000000]   node   0: [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] Initmem setup node 0 [mem 0x0000000040000000-0x000000007fffffff]
[    0.000000] On node 0 totalpages: 262144
[    0.000000]   Normal zone: 1728 pages used for memmap
[    0.000000]   Normal zone: 0 pages reserved
[    0.000000]   Normal zone: 196608 pages, LIFO batch:63
[    0.000000]   HighMem zone: 65536 pages, LIFO batch:15
[    0.000000] psci: probing for conduit method from DT.
[    0.000000] psci: Using PSCI v0.1 Function IDs from DT
[    0.000000] percpu: Embedded 20 pages/cpu s50764 r8192 d22964 u81920
[    0.000000] pcpu-alloc: s50764 r8192 d22964 u81920 alloc=20*4096
[    0.000000] pcpu-alloc: [0] 0 [0] 1 [0] 2 [0] 3
[    0.000000] Built 1 zonelists, mobility grouping on.  Total pages: 260416
[    0.000000] Kernel command line: root=UUID=8b839794-b689-4eae-9453-e6f0d462715a rootwait rootfstype=ext4 console=ttyS0,115200 console=tty1 hdmi.audio=EDID:0 disp.screen0_output_mode=1920x1080p60 consoleblank=0 loglevel=1 ubootpart=10c3bddc-01 ubootsource=mmc usb-storage.quirks=0x2537:0x1066:u,0x2537:0x1068:u   sunxi_ve_mem_reserve=0 sunxi_g2d_mem_reserve=0 sunxi_fb_mem_reserve=16 cgroup_enable=memory swapaccount=1
[    0.000000] Dentry cache hash table entries: 131072 (order: 7, 524288 bytes, linear)
[    0.000000] Inode-cache hash table entries: 65536 (order: 6, 262144 bytes, linear)
[    0.000000] mem auto-init: stack:off, heap alloc:on, heap free:off
[    0.000000] Memory: 881196K/1048576K available (9216K kernel code, 1458K rwdata, 3220K rodata, 1024K init, 331K bss, 36308K reserved, 131072K cma-reserved, 131060K highmem)
[    0.000000] random: get_random_u32 called from __kmem_cache_create+0x1d/0x29c with crng_init=0
[... 390 lines cut ...]
[   26.918208] dwmac-sun8i 1c30000.ethernet eth0: No MAC Management Counters available
[   26.918217] dwmac-sun8i 1c30000.ethernet eth0: PTP not supported by HW
[   26.919164] dwmac-sun8i 1c30000.ethernet eth0: configuring for phy/mii link mode
[   26.921854] dwmac-sun8i 1c30000.ethernet eth0: Link is Up - 100Mbps/Full - flow control off
[   26.922173] IPv6: ADDRCONF(NETDEV_CHANGE): eth0: link becomes ready
[   33.783057] vcc3v0: disabling
[   33.783074] vcc5v0: disabling
[   33.783080] usb0-vbus: disabling
[ 1566.335831] usb 8-1: new full-speed USB device number 2 using ohci-platform
[ 1566.557863] usb 8-1: config 1 interface 0 altsetting 1 endpoint 0x6 has invalid wMaxPacketSize 0
[ 1566.557879] usb 8-1: config 1 interface 0 altsetting 2 endpoint 0x6 has invalid wMaxPacketSize 0
[ 1566.557888] usb 8-1: config 1 interface 0 altsetting 3 endpoint 0x6 has invalid wMaxPacketSize 0
[ 1566.557898] usb 8-1: config 1 interface 0 altsetting 4 endpoint 0x6 has invalid wMaxPacketSize 0
[ 1566.557908] usb 8-1: config 1 interface 0 altsetting 5 endpoint 0x6 has invalid wMaxPacketSize 0
[ 1566.557918] usb 8-1: config 1 interface 0 altsetting 6 endpoint 0x6 has invalid wMaxPacketSize 0
[ 1566.557928] usb 8-1: config 1 interface 0 altsetting 7 endpoint 0x6 has invalid wMaxPacketSize 0
[ 1566.557938] usb 8-1: config 1 interface 0 altsetting 8 endpoint 0x6 has invalid wMaxPacketSize 0
[ 1566.557958] usb 8-1: New USB device found, idVendor=093a, idProduct=2625, bcdDevice= 1.00
[ 1566.557965] usb 8-1: New USB device strings: Mfr=0, Product=0, SerialNumber=0
[ 1566.941054] gspca_main: v2.14.0 registered
[ 1566.997445] gspca_main: gspca_pac7302-2.14.0 probing 093a:2625
[ 1567.004114] input: gspca_pac7302 as /devices/platform/soc/1c1d400.usb/usb8/8-1/input/input2
[ 1567.005506] usbcore: registered new interface driver gspca_pac7302
[ 1567.211619] usbcore: registered new interface driver snd-usb-audio
`

jacksonliam, 2021-04-29:
What's the output of `v4l2-ctl --list-formats-ext`?

miko671995, 2021-04-29:
root@orangepipc:/tmp/mjpg-streamer/mjpg-streamer-experimental# v4l2-ctl --list-formats-ext
ioctl: VIDIOC_ENUM_FMT
        Type: Video Capture

        [0]: 'ST12' (Sunxi Tiled NV12 Format, compressed)
        [1]: 'NV12' (Y/CbCr 4:2:0)

jacksonliam, 2021-04-29:
Yeah that camera is not going to work

closed 2021-04-29 by jacksonliam

tneijsel, 2023-02-22:
This camera works fine wit Webcamoid  on Desktop on XFCE on Orange pi Pc Plus running:

NAME="Ubuntu"
VERSION="20.04.5 LTS (Focal Fossa)"
ID=ubuntu
ID_LIKE=debian
PRETTY_NAME="Armbian 22.11.4 Focal"
VERSION_ID="20.04"
HOME_URL="https://www.ubuntu.com/"
SUPPORT_URL="https://help.ubuntu.com/"
BUG_REPORT_URL="https://bugs.launchpad.net/ubuntu/"
PRIVACY_POLICY_URL="https://www.ubuntu.com/legal/terms-and-policies/privacy-policy"
VERSION_CODENAME=focal
UBUNTU_CODENAME=focal

But I am unable to get it working. The purpose is to get the usb webcam image streaming in Octoprint which is installad and running.
in Motion the video I get unable to decode video.. 

V4l2 output:
v4l2-ctl --list devices
cedrus (platform:cedrus):
	/dev/video0
	/dev/media0

USB PHY 2.0: USB CAMERA (usb-1c1b000.usb-1):
	/dev/video1
	/dev/video2
	/dev/media1

And on format:
v4l2-ctl --list-formats
ioctl: VIDIOC_ENUM_FMT
	Type: Video Capture

	[0]: 'ST12' (Sunxi Tiled NV12 Format, compressed)
	[1]: 'NV12' (Y/CbCr 4:2:0)

So if the webcamoid application is running the camera on esktop by default is it not possible to get any image in Motion or any other program I tested. 

Help Appreciated.
Ton.