jacksonliam/mjpg-streamer #313: No output. Connection error

phodina, 2021-07-25:
Hi,
I start the `mjpeg-streamer` on my PC running Linux and I want to connect to the URI on the same machine. However, for some reason I get this error as an output:
```
501: Not Implemented!
no www-folder configured
```

Here's the command I entered:
```
$ mjpg_streamer -i input_uvc.so -o output_http.so
MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
```

Any idea what I do wrong?

lisowskibraeden, 2022-03-19:
Looks like you aren't enabling the www-folder-path

`mjpg_streamer -i input_uvc.so -o output_http.so -w /WHEREVER/THEY/ARE`

Look for html files someone in the project and change the above to point to their location.

FlorianCassayre, 2023-07-29:
> Looks like you aren't enabling the www-folder-path
> 
> `mjpg_streamer -i input_uvc.so -o output_http.so -w /WHEREVER/THEY/ARE`
> 
> Look for html files someone in the project and change the above to point to their location.

Correction: `mjpg_streamer -i input_uvc.so -o "output_http.so -w /WHEREVER/THEY/ARE"`

DangerManM, 2024-11-13:
First, "make clean && make uninstall", then, "make && make install", and finally, remember to add the "-y" command to the start.sh