jacksonliam/mjpg-streamer #314: date and time wrong in the stream timestamp

anzbert, 2021-08-04:
I am using this great tool to stream from a raspberry pi zero W with a usb webcam. i am using the input_uvc.so plugin and the output_http.so. Everything works fantastic, except I have no control over the on-stream timestamp, which is also showing the wrong time.
The linux date has been set correctly and the date function is showing the right time in the terminal.
I am unsure what I can do and if it is a UVC driver issue or a MJPG problem.

[image]

Thanks for the help!
Andreas

anzbert, 2021-08-04:
these are my startup settings:
mjpg_streamer -i "input_uvc.so --no_dynctrl -r 1280x720" -o "output_http.so -p 8080"

jacksonliam, 2021-08-04:
This is not added by mjpg-streamer so must be the camera or output device.

closed 2021-08-04 by jacksonliam