jacksonliam/mjpg-streamer #327: mjpeg streamer is getting slow while operating on  AR9331 based MCU

Ritesh236, 2021-11-16:
Hello Team,

We are using AR9331 module and also using mjpg-streamer package to stream video as per requirement of our product to stream live video streaming data over Wi-Fi Interface

**MJPG Streamer Version: 2.0**

Command to Stream video :- 

> mjpg_streamer -i "input_uvc.so -d /dev/video0 -r 1280x720 -f 30 -l auto -n" -o "output_http.so -p 8080 -w /www/webcam/"

When I am using command as mentioned above the video is lagging in the webpage as well as into Mobile side 

Can someone please let me know how can I increase the speed of video stream
Lagging means, when video is streaming then if I move an object in front of camera then on video its visible after sometime on the webpage(approx 1-3 seconds) and sometime it gets struct at some point of time.

Webpage URL -> <IP>:8080/?action=stream

Camera used is endoscope USB camera. Part number of Camera is HD603, provided by shenzhen angell technology co. ltd..

Linux kernel details is as mentioned below:
Image name: MIPS OpenWrt Linux-4.14.221
Build date: 2021-02-15 15:22:37 UTC
Architecture: MIPS
OS/image type: Linux Kernel
Compression: LZMA
Data size: 1.4 MB (1500591 bytes)

Linux version 4.14.221 (sysadmin@BuildDev-02) (gcc version 7.5.0 (OpenWrt GCC 7.5.0 r11306-c4a6851c72)) #0 Mon Feb 15 15:22:37 2021

Below are few AR9331 Characteristics for AR9331 Module
- CPU: AR9331 400Mhz MIPS core
- RAM: 64M DDR2 RAM
- Flash: 16M SPI flash (4/8/32m option)
- Wireless speed: 135Mbps
- General GPIO: 21 (except TX, RX)
- USB: Usb 2.0 master interface, support USB hub extension
- Power supply voltage: 3.3V.
- Port: 1WAN and 1LAN 10/100Mbps network interface
- Antenna: the built-in PCB /IPX external antenna.
- Debug: serial debugging interface has been out.
- Board Power: 0.36W
- Product size: 27.5 * 51.4MM +- 0.1mm
- SPI interface pin out

Please find below image regarding overall processes which are running into our device

[image]

Let me know if need anything else from our end

Regards,
Ritesh Prajapati

jacksonliam, 2021-11-16:
In this mode mjpg-streamer will be receiving JPEG frames from the camera and transferring them over the network as soon as it can.

The bottleneck in latency will be a combination of the time the camera takes to encode the frames and transfer them, the time your mjpg-streamer server takes to transfer those frames over the network,l (WiFi can be bad for this), or the time the client takes to display them.

I'd make sure the client is not buffering the frames before displaying them.

The best case I have seen in latency is something like 800ms, that's just the nature of using jpeg for video.

But otherwise, there are no changes you can make to mjpg-streamer itself to improve latency.

Cheers,
Liam.

closed 2021-11-16 by jacksonliam

Ritesh236, 2021-11-16:
@jacksonliam ,

Thanks for your prompt response.

Would you please suggest us like how it can be improved even if I want to use Wi-Fi as we have specific requirements to stream video over Wi-Fi?

Also we have already checked with Ethernet with same module but not seen much difference like it was still lagging at that time.

Regards,
Ritesh Prajapati

Ritesh236, 2021-11-22:
@jacksonliam ,

We have also compared with Orange PI in which below are major difference between comparison of AR9331 and Orange PI Module

- AR9331 Module has 64 MB RAM while Orange PI has 2GB of RAM
- AR9331 Module has 400 MHz CPU Clock while Orange PI has 480 MHz CPU Clock
- AR9331 Module has 16 MB Flash Memory while Orange PI has 
- Ar9331 Module has one CPU Core while Orange PI has 4 CPU Core

So, As per testing results Video Streaming is working fine without any issue into Orange PI board while facing lagging issues into AR9331 Module

We are running OpenWRT based firmware into both modules.

Do you have any idea like what can be the issue with AR9331 module having that much of hardware resource? It may be due to virtual memory which are not getting sufficient into AR9331 module? 

We stuck into critical state like already product is designed and facing this type of critical issues while doing video streaming.

Regards,
Ritesh Prajapati

Ritesh236, 2021-11-23:
@jacksonliam ,

Do you have any update regarding issue as per details which I have shared into last post? is it due to single core CPU or something else? 

Can you please help us for the same because we are  completely stuck as we checked with different resolution and didn't find exact root-cause like where exactly issue is?

Regards,
Ritesh Prajapati

jacksonliam, 2021-11-23:
The CPU should be powerful enough, as you say top is not showing high CPU usage.

Check if you can capture frames fast enough with something else that isn't mjpg-streamer. E.g. libav / ffmpeg.

Ritesh236, 2021-11-23:
@jacksonliam ,

We have also checked with ffmpeg and seems like getting 12 to 15 frames per seconds in case of 30 FPS for 1280 X 720 Resolution. 

I don;t think that issue is into camera driver because same camera with same driver is working fine into Orange PI board.

So, I don't have much idea like where actually issue is now to resolve it

Regards,
Ritesh Prajapati

Ritesh236, 2021-11-23:
@jacksonliam ,

Do you have any suggestions to work mjpeg streamer with AR9331 Module which has good resources in terms of CPU and Memory? 

Will it be possible at your end to verify it or can you please help us to resolve that video streaming slow down issue?

Regards,
Ritesh Prajapati