jacksonliam/mjpg-streamer #328: Streamer is getting Slow in AR9331 board

dhawal2611, 2021-11-18:
Hi all,

Board : AR9331 openwrt module

We are live video streaming over Wi-Fi Interface

I have Cross compiled mjpg-streamer package for OpenWRT system. Below are the steps for cross compiled mjpg-sreamer for AR9331 module

PATH=$PATH:/home/sysadmin/Public/Dhawal/openwrt_dhawal/vet-openwrt/openwrt-19.07.7/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/bin
export STAGING_DIR=/home/sysadmin/Public/Dhawal/openwrt_dhawal/vet-openwrt/openwrt-19.07.7/staging_dir/
export CROSS_COMPILE=mips-openwrt-linux-
export ARCH=mips
export CC=mips-openwrt-linux-gcc
make 

While Running the binary in the OpenWRT the device shows low speed.
At Resolution 1280x720 approx 7-8 frames are transferred per second
But At Resolution 640x480 approx 12-13 frames are transferred per second
Also At Resolution 320x240 approx 12-13 frames are transferred per second
The details is as shown in below image.

[image]

The frames are captured, but the count of frames at resolution 1280x720 is very low.
So, When transfering the image over http(seen video on below mentioned IP) gets slow down or struct sometime

<Device_IP>:8080/?action=stream

Camera used is endoscope USB camera. Part number of Camera is HD603, provided by shenzhen angell technology co. ltd..

So, Can I get the reason why the stream is slow.

dhawal2611, 2021-11-19:
Hi all, 

************************************************************************************************
Below are few more details regarding AR9331 board

BOARD: GainStrong Oolite v1-Dev
SOC: AR9330 rev. 1
CPU: MIPS 24Kc
RAM: 64 MB DDR2 16-bit CL3-4-4-10
FLASH: 16 MB Winbond W25Q128
MAC: 00:03:7F:09:0B:AD
CLOCKS: CPU/RAM/AHB/SPI/REF
               400/400/200/ 25/ 25 MHz

Hit any key to stop booting:  0
Booting image from 0x9F020000...

Image name:    MIPS OpenWrt Linux-4.14.221
Build date:    2021-02-15 15:22:37 UTC
Architecture:  MIPS
OS/image type: Linux Kernel
Compression:   LZMA
Data size:     1.4 MB (1500591 bytes)
Load address:  0x80060000
Entry point:   0x80060000

Number of CPU used : 1

CPU details are : 
root@SLSVETAIE:/# cat /proc/cpuinfo
system type             : Atheros AR9330 rev 1
machine                 : GainStrong Oolite V1.0
processor               : 0
cpu model               : MIPS 24Kc V7.4
BogoMIPS                : 265.42
wait instruction        : yes
microsecond timers      : yes
tlb_entries             : 16
extra interrupt vector  : yes
hardware watchpoint     : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa                     : mips1 mips2 mips32r1 mips32r2
ASEs implemented        : mips16
Options implemented     : tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit nan_legacy nan_2008 perf
shadow register sets    : 1
kscratch registers      : 0
package                 : 0
core                    : 0
VCED exceptions         : not available
VCEI exceptions         : not available

************************************************************************************************
In comparison below are details of orange Pi 3
CPU:   Allwinner H6 (SUN50I)
Model: OrangePi 3
DRAM:  2 GiB
MMC:   mmc@4020000: 0, mmc@4022000: 1
Loading Environment from FAT... Unable to use mmc 1:1... In:    serial@5000000
Out:   serial@5000000
Err:   serial@5000000
Net:   No ethernet found.

root@orangepi3:/home/tempie_fr# cat /proc/cpuinfo 
processor	: 0
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
Processor	: AArch64 Processor rev 4 (aarch64)
Hardware	: sun50iw1p1
BogoMIPS	: 48.00
Features	: fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer	: 0x41
CPU architecture: 8
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4
************************************************************************************************
Above details are for comparasion of the board regarding CPU and memory related details
Please Have a look on it and let me know why stream is slow in AR9331 module as compare to Orange Pi 3

Thanks in advance
Dhawal Lad

jacksonliam, 2021-11-19:
If you are using output file it could be because the storage is too slow to take more frames.
Or could be the amount of data the CPU can transfer over WiFi (WiFi sometimes is CPU heavy, seen this on the Pi with some WiFi adapters).
Could be something else running on the system using the CPU, or that the driver for the specific webcam you have is CPU heavy.

But tbh we've had mjpg-streamer running fine on hardware slower than this... Old ddwrt routers with 200Mhz and 32Mb RAM etc.

The problem will most likely not be with mjpg-streamer code but something to do with your environment.

closed 2021-11-19 by jacksonliam