jacksonliam/mjpg-streamer #329: Stream is Slow in AR9331 board

dhawal2611, 2021-11-19:
Hi all,

Board : AR9331 openwrt module

I have Cross compiled mjpg-streamer package for OpenWRT system. Below are the steps for cross compiled mjpg-sreamer for AR9331 module

PATH=$PATH:/home/sysadmin/Public/Dhawal/openwrt_dhawal/vet-openwrt/openwrt-19.07.7/staging_dir/toolchain-mips_24kc_gcc-7.5.0_musl/bin
export STAGING_DIR=/home/sysadmin/Public/Dhawal/openwrt_dhawal/vet-openwrt/openwrt-19.07.7/staging_dir/
export CROSS_COMPILE=mips-openwrt-linux-
export ARCH=mips
export CC=mips-openwrt-linux-gcc
make

While Running the binary in the OpenWRT the device shows **low speed**.
**At Resolution 1280x720 approx 7-8 frames are captured per second
But At Resolution 640x480 approx 12-13 frames are captured per second
Also At Resolution 320x240 approx 12-13 frames are captured per second**
The details is as shown in below image.

[image]

We have disabled code in output_file.so, So the image is not stored in file.

**_Just we are taking the frame from uvc only
That's the reason we have used the command mentioned_**

The frames are captured, but the count of frames at resolution 1280x720 is very low.
So, When transfering the image over http(seen video on below mentioned IP) gets slow down or struct sometime

**Camera used is endoscope USB camera. Part number of Camera is HD603, provided by shenzhen angell technology co. ltd..**

I have tested it on **Orange Pi 3,** the **stream is smooth** on it.

**So, Can I get the reason why the stream is slow on AR9331 board**

**************************************************************************************************************
Below are few more details regarding AR9331 board

BOARD: GainStrong Oolite v1-Dev
SOC: AR9330 rev. 1
CPU: MIPS 24Kc
RAM: 64 MB DDR2 16-bit CL3-4-4-10
FLASH: 16 MB Winbond W25Q128
MAC: 00:03:7F:09:0B:AD
CLOCKS: CPU/RAM/AHB/SPI/REF
400/400/200/ 25/ 25 MHz

Hit any key to stop booting: 0
Booting image from 0x9F020000...

Image name: MIPS OpenWrt Linux-4.14.221
Build date: 2021-02-15 15:22:37 UTC
Architecture: MIPS
OS/image type: Linux Kernel
Compression: LZMA
Data size: 1.4 MB (1500591 bytes)
Load address: 0x80060000
Entry point: 0x80060000

Number of CPU used : 1

CPU details are :
root@SLSVETAIE:/# cat /proc/cpuinfo
system type : Atheros AR9330 rev 1
machine : GainStrong Oolite V1.0
processor : 0
cpu model : MIPS 24Kc V7.4
BogoMIPS : 265.42
wait instruction : yes
microsecond timers : yes
tlb_entries : 16
extra interrupt vector : yes
hardware watchpoint : yes, count: 4, address/irw mask: [0x0ffc, 0x0ffc, 0x0ffb, 0x0ffb]
isa : mips1 mips2 mips32r1 mips32r2
ASEs implemented : mips16
Options implemented : tlb 4kex 4k_cache prefetch mcheck ejtag llsc dc_aliases perf_cntr_intr_bit nan_legacy nan_2008 perf
shadow register sets : 1
kscratch registers : 0
package : 0
core : 0
VCED exceptions : not available
VCEI exceptions : not available

**************************************************************************************************************
In comparison below are details of orange Pi 3

CPU: Allwinner H6 (SUN50I)
Model: OrangePi 3
DRAM: 2 GiB
MMC: mmc@4020000: 0, mmc@4022000: 1
Loading Environment from FAT... Unable to use mmc 1:1... In: serial@5000000
Out: serial@5000000
Err: serial@5000000
Net: No ethernet found.

root@orangepi3:/home/tempie_fr# cat /proc/cpuinfo
processor : 0
Processor : AArch64 Processor rev 4 (aarch64)
Hardware : sun50iw1p1
BogoMIPS : 48.00
Features : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 1
Processor : AArch64 Processor rev 4 (aarch64)
Hardware : sun50iw1p1
BogoMIPS : 48.00
Features : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 2
Processor : AArch64 Processor rev 4 (aarch64)
Hardware : sun50iw1p1
BogoMIPS : 48.00
Features : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4

processor : 3
Processor : AArch64 Processor rev 4 (aarch64)
Hardware : sun50iw1p1
BogoMIPS : 48.00
Features : fp asimd evtstrm aes pmull sha1 sha2 crc32 cpuid
CPU implementer : 0x41
CPU architecture: 8
CPU variant : 0x0
CPU part : 0xd03
CPU revision : 4
**************************************************************************************************************

Above details are for comparison of the board regarding CPU and memory related details
**_Please Have a look on it and let me know why stream is slow in AR9331 module as compare to Orange Pi 3_**

Thanks in advance
Dhawal Lad

dhawal2611, 2021-11-19:
Below Image shows frames read from camera per second at resolution 1280x720
[image]
Below Image shows frames read from camera per second at resolution 640x480
[image]
Below Image shows frames read from camera per second at resolution 320x240
[image]

Please let me know why we are getting slow stream and less frames in the device
**_NOTE : "output_file.so" this file dose not contain any code, No files are generated_**

jacksonliam, 2021-11-19:
Closed duplicate #328.

Please don't open a new issue for the same thing. This is not an issue with mjpg-streamer code. You can reply to the closed issue.

Check top command, what is using all the CPU?

Are you able to capture frames faster with any other software on this CPU?

closed 2021-11-19 by jacksonliam

dhawal2611, 2021-11-19:
[image]

Please have a look on the above image. It shows that Nothing is using CPU
So What may be the reason??????

jacksonliam, 2021-11-19:
Possibly USB bandwidth issue, are you able to grab frames faster with another application e.g. libav / ffmepg?

dhawal2611, 2021-11-19:
lsusb logs
[image]

Below are the logs that shows the high speed USB detected

************************************************************************************************

root@SLSVETAIE:/# lsusb -d 1d6b:0002 -v

Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
Device Descriptor:
  bLength                18
  bDescriptorType         1
  bcdUSB               2.00
  bDeviceClass            9 Hub
  bDeviceSubClass         0 Unused
  bDeviceProtocol         1 Single TT
  bMaxPacketSize0        64
  idVendor           0x1d6b Linux Foundation
  idProduct          0x0002 2.0 root hub
  bcdDevice            4.14
  iManufacturer           3 Linux 4.14.221 ehci_hcd
  iProduct                2 EHCI Host Controller
  iSerial                 1 ehci-platform
  bNumConfigurations      1
  Configuration Descriptor:
    bLength                 9
    bDescriptorType         2
    wTotalLength           25
    bNumInterfaces          1
    bConfigurationValue     1
    iConfiguration          0
    bmAttributes         0xe0
      Self Powered
      Remote Wakeup
    MaxPower                0mA
    Interface Descriptor:
      bLength                 9
      bDescriptorType         4
      bInterfaceNumber        0
      bAlternateSetting       0
      bNumEndpoints           1
      bInterfaceClass         9 Hub
      bInterfaceSubClass      0 Unused
      bInterfaceProtocol      0 Full speed (or root) hub
      iInterface              0
      Endpoint Descriptor:
        bLength                 7
        bDescriptorType         5
        bEndpointAddress     0x81  EP 1 IN
        bmAttributes            3
          Transfer Type            Interrupt
          Synch Type               None
          Usage Type               Data
        wMaxPacketSize     0x0004  1x 4 bytes
        bInterval              12
Hub Descriptor:
  bLength               9
  bDescriptorType      41
  nNbrPorts             1
  wHubCharacteristic 0x0009
    Per-port power switching
    Per-port overcurrent protection
    TT think time 8 FS bits
  bPwrOn2PwrGood       10 * 2 milli seconds
  bHubContrCurrent      0 milli Ampere
  DeviceRemovable    0x00
  PortPwrCtrlMask    0xff
 Hub Port Status:
   Port 1: 0000.0503 highspeed power enable connect
Device Status:     0x0001
  Self Powered

************************************************************************************************
USB detected LOGS

[20740.620719] usb 1-1: new high-speed USB device number 2 using ehci-platform
[20740.859462] uvcvideo: Found UVC 1.00 device USB Camera (0bda:5846)
[20740.876413] input: USB Camera: USB Camera as /devices/platform/ehci-platform/usb1/1-1/1-1:1.0/input/input0

dhawal2611, 2021-11-25:
Hi, 
I have tested the code with ffmpeg but when I tried to run the code with Resolution with 1280*720 then crash is occured and code gets stop due to low memory while running the ffmpeg code

Running ffmpeg with resolution 640*480 or 320*240 the code gets executed and we get the frames count between 12-13 frames/sec

**************************************************************************************************************************************

Logs at Resolution 640*480

root@SLSVETAIE:/tmp# ffmpeg -f v4l2 -input_format mjpeg -framerate 30 -video_siz
e 640x480 -i /dev/video0 -c:v copy output.mkv[Jv[J[J[J[J[J[J[J[J[J[J[J[J[J[J[J[J[J[J[J[J-f mpegts -flush_packets 0 udp://192.168.1.4:5000?pkt_s

ize=1316
ffmpeg version 3.4.7 Copyright (c) 2000-2019 the FFmpeg developers
  built with gcc 7.5.0 (OpenWrt GCC 7.5.0 r11306-c4a6851c72)
  configuration: --enable-cross-compile --cross-prefix=mips-openwrt-linux-musl- --arch=mips --cpu=24kc --target-os=linux --prefix=/usr --pkg-config=pkg-config --enable-shared --enable-static --enable-pthreads --enable-zlib --disable-doc --disable-debug --disable-lzma --disable-vaapi --disable-vdpau --disable-outdevs --disable-altivec --disable-vsx --disable-power8 --disable-armv5te --disable-armv6 --disable-armv6t2 --disable-inline-asm --disable-mipsdsp --disable-mipsdspr2 --disable-mipsfpu --disable-msa --disable-mmi --disable-fast-unaligned --disable-runtime-cpudetect --disable-asm --disable-inline-asm --disable-x86asm --enable-avresample --disable-decoder=atrac3 --disable-decoder=h264 --disable-decoder=hevc --disable-decoder=vc1 --disable-muxer=h264 --disable-muxer=hevc --disable-muxer=vc1 --disable-demuxer=h264 --disable-demuxer=hevc --disable-demuxer=vc1 --disable-parser=h264 --disable-parser=hevc --disable-parser=vc1 --enable-small --enable-libshine
  libavutil      55. 78.100 / 55. 78.100
  libavcodec     57.107.100 / 57.107.100
  libavformat    57. 83.100 / 57. 83.100
  libavdevice    57. 10.100 / 57. 10.100
  libavfilter     6.107.100 /  6.107.100
  libavresample   3.  7.  0 /  3.  7.  0
  libswscale      4.  8.100 /  4.  8.100
  libswresample   2.  9.100 /  2.  9.100
[0;35m[video4linux2,v4l2 @ 0x77ecc710] [0mThe driver changed the time per frame from 1/30 to 1/15
Input #0, video4linux2,v4l2, from '/dev/video0':
  Duration: N/A, start: 456.961099, bitrate: N/A
    Stream #0:0: Video: mjpeg, yuvj422p(pc, bt470bg/unknown/unknown), 640x480, 15 fps, 15 tbr, 1000k tbn, 1000k tbc
Stream mapping:
  Stream #0:0 -> #0:0 (mjpeg (native) -> mpeg2video (native))
Press [q] to stop, [?] for help
[1;34m[swscaler @ 0x67c000] [0m[0;33mdeprecated pixel format used, make sure you did set range correctly
[0mOutput #0, mpegts, to 'udp://192.168.1.4:5000?pkt_size=1316':
  Metadata:
    encoder         : Lavf57.83.100
    Stream #0:0: Video: mpeg2video, yuv422p, 640x480, q=2-31, 200 kb/s, 15 fps, 90k tbn, 15 tbc
    Metadata:
      encoder         : Lavc57.107.100 mpeg2video
    Side data:
      cpb: bitrate max/min/avg: 0/0/200000 buffer size: 0 vbv_delay: -1
frame=    2 fps=0.6 q=4.7 size=      26kB time=-00:00:00.13 bitrate=N/A speed=N/A    
frame=    3 fps=0.6 q=2.0 size=      66kB time=00:00:00.00 bitrate=48811636.4kbits/s speed=2.02e-06x    
frame=    4 fps=0.5 q=2.0 size=     112kB time=00:00:00.13 bitrate=6869.0kbits/s speed=0.0181x    
frame=    5 fps=0.6 q=2.0 size=     154kB time=00:00:00.20 bitrate=6316.5kbits/s speed=0.0235x    
frame=    6 fps=0.6 q=3.6 size=     171kB time=00:00:00.26 bitrate=5250.6kbits/s speed=0.0279x    
frame=    7 fps=0.7 q=6.0 size=     179kB time=00:00:00.33 bitrate=4390.0kbits/s speed=0.031x    
frame=    8 fps=0.7 q=8.7 size=     182kB time=00:00:00.40 bitrate=3737.3kbits/s speed=0.0333x    
frame=    9 fps=0.7 q=11.5 size=     185kB time=00:00:00.46 bitrate=3248.6kbits/s speed=0.0367x    
frame=   10 fps=0.7 q=14.4 size=     185kB time=00:00:00.60 bitrate=2526.7kbits/s speed=0.0447x    
frame=   11 fps=0.8 q=15.1 size=     186kB time=00:00:00.73 bitrate=2081.6kbits/s speed=0.0513x    
frame=   12 fps=0.8 q=14.2 size=     188kB time=00:00:00.80 bitrate=1921.3kbits/s speed=0.0534x    
frame=   13 fps=0.8 q=13.2 size=     189kB time=00:00:00.86 bitrate=1785.7kbits/s speed=0.0554x    
frame=   14 fps=0.9 q=9.7 size=     207kB time=00:00:00.93 bitrate=1816.1kbits/s speed=0.0572x    
frame=   15 fps=0.9 q=12.9 size=     208kB time=00:00:01.00 bitrate=1705.5kbits/s speed=0.0587x    
frame=   16 fps=0.9 q=13.5 size=     209kB time=00:00:01.06 bitrate=1608.8kbits/s speed=0.0601x    
frame=   17 fps=0.9 q=13.8 size=     211kB time=00:00:01.13 bitrate=1523.4kbits/s speed=0.0615x    
frame=   18 fps=0.9 q=13.3 size=     212kB time=00:00:01.20 bitrate=1447.6kbits/s speed=0.0628x    
frame=   19 fps=1.0 q=13.2 size=     213kB time=00:00:01.26 bitrate=1379.7kbits/s speed=0.064x    
frame=   20 fps=1.0 q=13.0 size=     215kB time=00:00:01.33 bitrate=1318.6kbits/s speed=0.0651x    
frame=   21 fps=1.0 q=12.8 size=     216kB time=00:00:01.40 bitrate=1263.4kbits/s speed=0.0661x    
frame=   22 fps=1.0 q=12.5 size=     217kB time=00:00:01.46 bitrate=1213.1kbits/s speed=0.0671x    
frame=   23 fps=1.0 q=12.3 size=     218kB time=00:00:01.53 bitrate=1167.2kbits/s speed=0.068x    
frame=   24 fps=1.0 q=12.8 size=     220kB time=00:00:01.66 bitrate=1080.2kbits/s speed=0.0718x    
frame=   25 fps=1.0 q=12.6 size=     221kB time=00:00:01.73 bitrate=1044.7kbits/s speed=0.0725x    
[... cut ...]
[  100.591705] wlan0: associate with ec:bd:1d:c3:90:b0 (try 1/3)
[  100.598937] wlan0: RX AssocResp from ec:bd:1d:c3:90:b0 (capab=0x431 status=0 aid=108)
[  100.605775] wlan0: associated

root@SLSVETAIE:/# 
root@SLSVETAIE:/# [H[J
root@SLSVETAIE:/# [J
root@SLSVETAIE:/# 

**************************************************************************************************************************************

So what can we do to get more number of frames at resolution 1280*720 with mjpg-streamer in AR9331 module.

Thanks & Regards

jacksonliam, 2021-11-25:
If ffmpeg captures frames at 640x480 at the same rate as mjpg-streamer then you know the user space software is not the issue. This is not an mjpg-streamer issue. It is likely an issue with your hardware/drivers/OS.

dhawal2611, 2021-12-03:
Ok, I am looking into it and If I get something I will get back to you
Thanks for the support