jacksonliam/mjpg-streamer #330: the browser is turned on with no video stream output

xuganlai, 2021-11-28:
make and then run start.sh，browser url enter http://192.168.33.6:8080/stream.html,but no stream output.
1、./start.sh print:
MJPG Streamer Version: git rev: 310b29f4a94c46652b20c4b7b6e5cf24e532af39
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: 5
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
 i: FPS coerced ......: from 5 to 30
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: ./www/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: enabled
 i: select() timeout
 i: cleaning up resources allocated by input thread

2、about cameral:
1) v4l2-ctl -d /dev/video0 --all
Driver Info (not using libv4l2):
        Driver name   : uvcvideo
        Card type     : Integrated Camera
        Bus info      : usb-0000:02:03.0-1
        Driver version: 4.4.228
        Capabilities  : 0x84200001
                Video Capture
                Streaming
                Extended Pix Format
                Device Capabilities
        Device Caps   : 0x04200001
                Video Capture
                Streaming
                Extended Pix Format
Priority: 2
Video input : 0 (Camera 1: ok)
Format Video Capture:
        Width/Height      : 1280/720
        Pixel Format      : 'MJPG'
        Field             : None
        Bytes per Line    : 0
        Size Image        : 1843200
        Colorspace        : sRGB
        Transfer Function : Default
        YCbCr Encoding    : Default
        Quantization      : Default
        Flags             : 
Crop Capability Video Capture:
        Bounds      : Left 0, Top 0, Width 1280, Height 720
        Default     : Left 0, Top 0, Width 1280, Height 720
        Pixel Aspect: 1/1
Selection: crop_default, Left 0, Top 0, Width 1280, Height 720
Selection: crop_bounds, Left 0, Top 0, Width 1280, Height 720

2) lsusb show below：
Bus 001 Device 002: ID 13d3:56a6 IMC Networks
3) dmesg: 
[  747.775114] media: Linux media interface: v0.10
[  747.800674] Linux video capture interface: v2.00
[  747.846943] uvcvideo: Found UVC 1.00 device Integrated Camera (13d3:56a6)
[  747.904045] input: Integrated Camera as /devices/pci0000:00/0000:00:11.0/0000:02:03.0/usb1/1-1/1-1:1.0/input/input6
[  747.904323] usbcore: registered new interface driver uvcvideo
[  747.904326] USB Video Class driver (1.1.1)

jacksonliam, 2021-11-28:
Have you modified start.sh? If so what args are you running with? It looks like you're using -y but your camera supports mjpg?

xuganlai, 2021-11-30:
> Have you modified start.sh? If so what args are you running with? It looks like you're using -y but your camera supports mjpg?
 
1、no change，running with "./mjpg_streamer -i "./input_uvc.so" -o "./output_http.so -w ./www""
2、Integrated Camera support uvc,so modify plugins\input_uvc\input_uvc.c,input_init function init with ：“int width = 640, height = 480, fps = 5, format = V4L2_PIX_FMT_YUYV, i;”

jacksonliam, 2021-11-30:
Modifying the code and just putting
format = V4L2_PIX_FMT_YUYV will of course break things...

closed 2021-11-30 by jacksonliam