jacksonliam/mjpg-streamer #332: Intermittent video stream with Logitech C922 on Ubuntu 21.10 on Raspberry Pi 3B+

staticanime, 2021-11-30:
Hi all

I'm having a bit of a strange issue with mjpg-streamer in conjunction with my Logitech C922 USB webcam on a Raspberry Pi 3B+ running Ubuntu 21.10. The "stream" from the webcam will randomly stutter/freeze from a few seconds up to minutes, and will occasionally result in no output at all

Been banging my head against the wall on this one, not sure why it's not working. I'm using the Pi to run Klipper/Mainsail for my 3d printer, and I've installed mjpg-streamer through KIAUH: https://github.com/th33xitus/kiauh

In my webcam.txt file I've specified
`camera="usb"
camera_usb_options="-r 1280x720 -f 30 -d /dev/v4l/by-id/usb-046d_C922_Pro_Stream_Webcam_95DE60EF-video-index0`

In the logs, this is what I see
`Starting up webcamDaemon...

--- Configuration: ----------------------------
cfg_file:      /home/pi/klipper_config/webcam.txt
camera:        usb
usb options:   -r 1280x720 -f 30 -d /dev/v4l/by-id/usb-046d_C922_Pro_Stream_Webcam_95DE60EF-video-index0 -timeout 30
raspi options: -fps 10
http options:  -w ./www-mjpgstreamer -n

Explicitly USB device: /dev/v4l/by-id/usb-046d_C922_Pro_Stream_Webcam_95DE60EF-video-index0
-----------------------------------------------

Found video devices:
/dev/video0
/dev/video1
/dev/video10
/dev/video11
/dev/video12
/dev/video13
/dev/video14
/dev/video15
/dev/video16
config file='/home/pi/klipper_config/webcam.txt':USB device was set in options and found in devices, start MJPG-streamer with the configured USB video device: /dev/video0
<13>Nov 30 10:50:45 pi: Starting USB webcam
Checking for VL805 (Raspberry Pi 4)...
  - It seems that you don't have VL805 (Raspberry Pi 4).
    There should be no problems with USB (a.k.a. select() timeout)
Running ./mjpg_streamer -o output_http.so -w ./www-mjpgstreamer -n -i input_uvc.so -r 1280x720 -f 30 -d /dev/video0
MJPG Streamer Version: git rev: 310b29f4a94c46652b20c4b7b6e5cf24e532af39
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 1280 x 720
 i: Frames Per Second.: 30
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: ./www-mjpgstreamer/
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: (null)
 o: username:password....: disabled
 o: commands.............: disabled
Done bring up all configured video device

Goodbye...

 i: select() timeout
 i: cleaning up resources allocated by input thread
`

I think the select() timeout is whats tripping me up, but I don't know where to go from here with further diagnostics

staticanime, 2021-11-30:
Took a screen capture of what I'm talking about: https://www.youtube.com/watch?v=zQQ-1460x_M

jacksonliam, 2021-11-30:
Yeah the select timeout is the issue if the stream goes down. I'm not sure about the stutter/freeze.

Are you sure the camera is getting enough power etc (e.g unpowered USB hubs). 
It's possible it's a system setup / hardware issue.

staticanime, 2021-11-30:
It's currently running directly off the Pi's USB ports, I have the webcam and my 3d printer board plugged in. Don't suppose you know of any way for me to check/confirm power draw to be the issue without buying a (powered) USB hub?

staticanime, 2021-11-30:
Found a way to check, seems to be requesting the std 500mA from the USB port, which the Pi should be well fit to handle?

```sudo lsusb -v|egrep "^Bus|MaxPower"
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
cannot read device status, Resource temporarily unavailable (11)
Bus 001 Device 005: ID 1d50:614e OpenMoko, Inc. stm32f103xe
    MaxPower              100mA
Bus 001 Device 004: ID 046d:085c Logitech, Inc. C922 Pro Stream Webcam
    MaxPower              500mA
can't get debug descriptor: Resource temporarily unavailable
Bus 001 Device 006: ID 0424:7800 Microchip Technology, Inc. (formerly SMSC)
    MaxPower                2mA
can't get debug descriptor: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable
Bus 001 Device 003: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
    MaxPower                2mA
Bus 001 Device 002: ID 0424:2514 Microchip Technology, Inc. (formerly SMSC) USB 2.0 Hub
can't get debug descriptor: Resource temporarily unavailable
    MaxPower                2mA
Bus 001 Device 001: ID 1d6b:0002 Linux Foundation 2.0 root hub
    MaxPower                0mA
can't get device qualifier: Resource temporarily unavailable
can't get debug descriptor: Resource temporarily unavailable```

jacksonliam, 2021-11-30:
Yeah it should be fine as long as the PSU powering the pi is good enough.

Best bet is to verify some other app can capture video correctly for the same amount of time. Because it's likely this is not an mjpg-streamer issue.

James4Ever0, 2024-09-08:
@staticanime 

Solution: https://github.com/jacksonliam/mjpg-streamer/issues/182#issuecomment-2336583229