jacksonliam/mjpg-streamer #334: Unable to select secondary webcam

biarkiw, 2021-12-04:
I am running Octoprint on a Ubuntu laptop, and I recently picked up a Logitech C270 to attach to my printer. Octoprint works great and I have gotten mjpg-streamer working, however whenever I attempt to start the stream it defaults to /dev/video0 even when I send the command  
`mjpg_streamer -i 'input_uvc.so [-d /dev/video2]' -o output_http.so` 

I get the output
`MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT`

`MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: (null)
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: init_VideoIn failed`

and when I put in the command
`mjpg_streamer -i 'input_uvc.so [-d /dev/video2]' -o output_http.so`

I get
`MJPG Streamer Version.: 2.0
 i: Using V4L2 device.: (null)
 i: Desired Resolution: 640 x 480
 i: Frames Per Second.: -1
 i: Format............: JPEG
 i: TV-Norm...........: DEFAULT
 i: init_VideoIn failed`

Any assistance in solving this issue would be much appreciated, I know my setup is abnormal, and therefore is likely more difficult to solve issues on.

jacksonliam, 2021-12-04:
The square brackets [ ] around the -d argument are wrong, remove those and it should work.

Why did you add them?

closed 2021-12-04 by jacksonliam

biarkiw, 2021-12-04:
The bracket's weren't the issue, as the second command wasn't supposed to have them in it and I must have copied the wrong command and not noticed. Somehow restarting my system before I start the camera every-time is making it work, and I no longer get the null error, unless I have had the camera active during this boot

biarkiw, 2021-12-04:
Also the reason I added them was due to the documentation, as I wasn't sure if they should be left or included