jacksonliam/mjpg-streamer #340: Transmitting MJPG over serial

bobby1321, 2022-02-01:
I'm currently working on a project where I want to transmit a video stream over UART serial on a pair of Pis. Currently, the transmission using mjpg-streamer works great, and I can send my video signal and see it on an oscilloscope, and I can save the serial data to a file using Python, but I can't use mjpg-streamer to read the video over serial. The way I am writing to serial is by treating the serial port as a "file", since everything is just a file on Linux. However, using the same method on the receiver doesn't work. 

Is there a way currently built into the library to decode this serial data in real time, or is this something completely unrealistic to expect? Please let me know if there is any other info needed. Thanks!

jacksonliam, 2022-02-02:
Thats an interesting use-case!
What options are you trying to use on the sending and receiving side? If you're using the -m option of `output_file` to output as a mjpeg file, into the serial port buffer, I don't think anything currently in mjpg-streamer would be able to receive it.

The mjpeg file is just a series of jpeg frames written to a file with no boundary or mark between them. `input_file` is designed more to watch a folder of individual jpeg files.

I guess on the recieving side you could pipe it into VLC or something which knows how to play an mjpg stream?

To be able to recieve such a file with mjpg-streamer, you'd have to modify input file to support streaming from a single file, it would have to decode the jpeg header enough to know when an individual frame ends so it can write just that single frame to the buffer.

bobby1321, 2022-02-07:
Sorry for taking so long to reply, I had to get my raspberry pi back from someone else.

On the transmitter side, I'm using the following options:
`mjpg_streamer -o "output_file.so -m serial0 -f /dev" -i "input_raspicam.so -x 640 -y 480 -fps 5" -o "output_http.so -p 8081" -b`
I have the pi transmitting over serial and http so that I can watch live what it is broadcasting.

For the receiver side, I have a python script which just takes the bitsream from the serial port and writes it to a file. This way I can look at the file after and see if it works. 
`import serial

ser = serial.Serial('/dev/serial0',921600)

f = open('test.jpeg','wb')

while 1 :
        f.write(ser.read(8))
`
It works sometimes. I'm not the most knowledgeable when it comes to writing code for this sort of thing, so any improvements are welcome. 

Piping into VLC on the receiving end is a good idea and I'll give that  a shot and let you know how it goes.
Thanks!