jacksonliam/mjpg-streamer #342: Not able to stop individual port steaming into mjpeg streamer

Ritesh236, 2022-02-17:
Hello Team,

We have started to stream video into 2 different ports 8080 & 8081 as per requirement using follwoing command over mjpeg streamer

> mjpg_streamer -i "input_uvc.so -d /dev/video0 -r 1280x720 -f 30 -l auto -n" -o "output_http.so -p 8080 -w /www/webcam/" -o "output_http.so -p 8081 -w /www/webcam/" &

Now, We have one customer requirement to stop video streaming on perticular port for few duration and then again need to start streaming into same port after that duration.

so, can it be possible into mjpeg streamer command or need to stop complete mjpeg streamer and then start again?

Ritesh236, 2022-02-18:
@jacksonliam ,

Do you have any update or any input regarding query which i have sent?

Let me know if need anything else from my end.

Regards,
Ritesh Prajapati

jacksonliam, 2022-02-18:
Hi,
This is not a feature, youd have to code it yourself to start and stop a specific plugin. 

I'd say run two instances of mjpg-streamer so you can restart the second one but that would only work if you used two different cameras.

I guess you could run a second instance of mjpg-streamer with input_http which connects to the first via localhost. I'm not sure how reliable that would be or how much delay and framedropping there would be on the second instance.

Cheers,
Liam.

closed 2022-02-18 by jacksonliam