jacksonliam/mjpg-streamer #349: libjpeg.so.8: cannot open shared object file: No such file or directory

AD17102, 2022-04-07:
I was compiling the mjpg-streamer for my raspberry pi camera when i got the error messages: 

ERROR: could not find input plugin
       Perhaps you want to adjust the search path with:
       # export LD_LIBRARY_PATH=/path/to/plugin/folder
       dlopen: libjpeg.so.8: cannot open shared object file: No such file or directory

when I tried to update the libjpeg8 package there was only libjpeg9 or libjpeg-62turbo
Any knowledge as to why?

duan-dt, 2022-04-27:
> I was compiling the mjpg-streamer for my raspberry pi camera when i got the error messages:
>
> ERROR: could not find input plugin Perhaps you want to adjust the search path with: # export [...]

Same issue here. Seems like the “input_raspicam” files are no longer compiled

jsaray, 2022-04-29:
I solved this by downloading the rpm here https://rpmfind.net/linux/rpm2html/search.php?query=libjpeg.so.8()(64bit)
and then installing it using rpm -i lib64jpeg8-2.1.2-4-omv4050.x86_64.rpm

duan-dt, 2022-04-30:
> I solved this by downloading the rpm here https://rpmfind.net/linux/rpm2html/search.php?query=libjpeg.so.8()(64bit)
> and then installing it using rpm -i lib64jpeg8-2.1.2-4-omv4050.x86_64.rpm

My workaround was through KIAUH and enabling legacy camera support. I am not entirely sure why it works because KIAUH also clones and builds the code here but at least I got it working.