jacksonliam/mjpg-streamer #365: Build of input_libcamera.so fails on Octopi / Raspbian 11

bryn51, 2022-10-06:
OK, I have Raspberry Pi 4B/8GB, which boots into Octopi based on latest nightly Build Raspian 11 / 64 Bit version.

whilst attempting to run the make for: mjpg-streamer-libcamera, one of the tasks is to build input_libcamera.so, essential to he overall role of this software to get video from the camera and stream it to http.

this is the output of the make:(partial, where the errors begin).
[ 55%] Building CXX object plugins/input_libcamera/CMakeFiles/input_libcamera.dir/input_libcamera.cpp.o
In file included from /home/pi/mjpg-streamer-libcamera/mjpg-streamer-experimental/plugins/input_libcamera/LibCamera.h:20,
                 from /home/pi/mjpg-streamer-libcamera/mjpg-streamer-experimental/plugins/input_libcamera/input_libcamera.cpp:31:
/usr/include/libcamera/libcamera/controls.h: In instantiation of ‘void libcamera::ControlList::set(const libcamera::Control<T>&, const std::initializer_list<_Up>&) [with T = libcamera::Span<const long long int, 2>; V = long long int]’:
/home/pi/mjpg-streamer-libcamera/mjpg-streamer-experimental/plugins/input_libcamera/input_libcamera.cpp:259:77:   required from here
/usr/include/libcamera/libcamera/controls.h:403:14: error: no matching function for call to ‘libcamera::ControlValue::set<libcamera::Span<const long long int, 2> >(libcamera::Span<const long long int, 4294967295>)’
  403 |   val->set<T>(Span<const typename std::remove_cv_t<V>>{ value.begin(), value.size() });
      |   ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/usr/include/libcamera/libcamera/controls.h:178:7: note: candidate: ‘template<class T, typename std::enable_if<((! libcamera::details::is_span<U>::value) && (! std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value)), std::nullptr_t>::type <anonymous> > void libcamera::ControlValue::set(const T&)’
  178 |  void set(const T &value)
      |       ^~~
/usr/include/libcamera/libcamera/controls.h:178:7: note:   template argument deduction/substitution failed:
/usr/include/libcamera/libcamera/controls.h:177:30: error: no type named ‘type’ in ‘struct std::enable_if<false, std::nullptr_t>’
  177 |            std::nullptr_t> = nullptr>
      |                              ^~~~~~~
/usr/include/libcamera/libcamera/controls.h:190:7: note: candidate: ‘template<class T, typename std::enable_if<(libcamera::details::is_span<U>::value || std::is_same<std::__cxx11::basic_string<char>, typename std::remove_cv< <template-parameter-1-1> >::type>::value), std::nullptr_t>::type <anonymous> > void libcamera::ControlValue::set(const T&)’
  190 |  void set(const T &value)
      |       ^~~
/usr/include/libcamera/libcamera/controls.h:190:7: note:   template argument deduction/substitution failed:
/usr/include/libcamera/libcamera/controls.h:403:14: note:   cannot convert ‘libcamera::Span<const long long int, 4294967295>((& value)->std::initializer_list<long long int>::begin(), (& value)->std::initializer_list<long long int>::size())’ (type ‘libcamera::Span<const long long int, 4294967295>’) to type ‘const libcamera::Span<const long long int, 2>&’
  403 |   val->set<T>(Span<const typename std::remove_cv_t<V>>{ value.begin(), value.size() });
      |   ~~~~~~~~~~~^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
make[3]: *** [plugins/input_libcamera/CMakeFiles/input_libcamera.dir/build.make:82: plugins/input_libcamera/CMakeFiles/input_libcamera.dir/input_libcamera.cpp.o] Error 1
make[3]: Leaving directory '/home/pi/mjpg-streamer-libcamera/mjpg-streamer-experimental/_build'
make[2]: *** [CMakeFiles/Makefile2:475: plugins/input_libcamera/CMakeFiles/input_libcamera.dir/all] Error 2
make[2]: Leaving directory '/home/pi/mjpg-streamer-libcamera/mjpg-streamer-experimental/_build'
make[1]: *** [Makefile:149: all] Error 2
make[1]: Leaving directory '/home/pi/mjpg-streamer-libcamera/mjpg-streamer-experimental/_build'
make: *** [Makefile:19: all] Error 2
==========

Clearly the make doesnt like 'controls.h' which is part of the libcamera installation.  I know insufficient about this environment to further anlayse, but clearly its causing enough of a problem to stop the process.
And without this my Arducam 16MP AF device cannot 'work' to do its job to provide vision into the octoprint application.
So, this is all supposed to work.  Please can someone point me to a stable version that works so I can move onward.  Or just fix this issue.  I have posted this same Issue onto the libcamera git.

LordBoos, 2022-10-07:
I have the exact same problem. 
Only workaround I have fond so far is to use prebuild image with Arducam 116MP AF support. Only bad thing about it is that it's older build from January so it's not the latest OctoPi nightly. But you can update OctoPi later from UI or SSH.
https://forum.arducam.com/t/aufocus-camera-with-octoprint-on-pi-zero/2558/3

khenriks, 2022-10-07:
I ran into the same issue too. This looks like it's due to a bug in libcamera that was just [fixed](https://git.libcamera.org/libcamera/libcamera.git/commit/?id=f9c28992a9f591124db24b44d94b88ff4ca294f8).

However, until the fix is released, this can be addressed by updating `plugins/input_libcamera/input_libcamera.cpp` line 259 from
```
controls_.set(controls::FrameDurationLimits, { frame_time, frame_time });
```
to
```cc
controls_.set(controls::FrameDurationLimits, Span<const int64_t, 2>({ frame_time, frame_time }));
```

I've also pushed this change to my own fork: [khenriks/mjpg-streamer](https://github.com/khenriks/mjpg-streamer).

LordBoos, 2022-10-07:
I managed to find another workaround before the fix is released.
If you downgrade your libcamera-dev and libcamera-apps to version 0.0.7 then the mjpg-streamer will compile. Versions 0.0.8 and 0.0.9 won't work.

bryn51, 2022-10-08:
Thanks guys for responding. At least I now know "it's not just me"!

I'll look into a) precompiled version of input-libcamera.so and b) going back a version to 0.0.7 
And see which works.

Later it may be possible to upgrade, but if the older/precompiled version delivers a result with octoprint I don't much care.
I can move on without the need to buy another camera.

Cheers!

bryn51, 2022-10-08:
> I have the exact same problem. Only workaround I have fond so far is to use prebuild image with Arducam 16MP AF support. Only bad thing about it is that it's older build from January so it's not the latest OctoPi nightly. But you can update OctoPi later from UI or SSH. https://forum.arducam.com/t/aufocus-camera-with-octoprint-on-pi-zero/2558/3

The fix suggested by @khenriks was effective.

jacksonliam, 2022-10-08:
> I ran into the same issue too. This looks like it's due to a bug in libcamera that was just [fixed](https://git.libcamera.org/libcamera/libcamera.git/commit/?id=f9c28992a9f591124db24b44d94b88ff4ca294f8).
> 
> However, until the fix is released, this can be addressed by updating `plugins/input_libcamera/input_libcamera.cpp` line 259 from
> 
> ```
> controls_.set(controls::FrameDurationLimits, { frame_time, frame_time });
> ```
> 
> to
> 
> ```c++
> controls_.set(controls::FrameDurationLimits, Span<const int64_t, 2>({ frame_time, frame_time }));
> ```
> 
> I've also pushed this change to my own fork: [khenriks/mjpg-streamer](https://github.com/khenriks/mjpg-streamer).

Thanks for this! Glad it's been resolved by libcamera

LordBoos, 2022-10-08:
This was now fixed in https://github.com/ArduCAM/mjpg-streamer/commit/a35c442a605f65444c01eb054656ed4a180c7cae
mjpg-streamer will compile with libcamera 0.0.9.

bryn51, 2022-10-08:
Yes. I verified the fix for mjpg_libcamera build (related to Arducam camera)

I applied the 3 changed lines of code to the controls.h on my system and  input_libcamera.so now compiles correctly. 

And the input_libcamera.so does its job too, The video image is acquired and passed through to Octoprint application.