jacksonliam/mjpg-streamer #366: V4L2 device.: (null) when starting on boot as a service

SaintTaint, 2022-10-13:
Currently setting up Octoprint on a Jetson Nano, using a USB webcam on /dev/video1.  Mjpg-streamer runs fine if i start it from the terminal with this command:  
`./mjpg_streamer`` -i "input_uvc.so -d /dev/video1" -o "output_http.so"`

When the same command runs from an init.d startup script, mjpg-streamer returns:
`MJPG-streamer [4003]: Using V4L2 device.: (null)`

When run from the terminal, "Using V4L2 device" returns /dev/video1 like it should.

I'm very confused.  Any ideas?

jacksonliam, 2022-10-14:
This is not an mjpg-streamer issue. As you said it works when running it from the terminal.

This is likely due to the user the startup script is running under not having permissions to access the /dev/video1 device. I think it has to be a member of the `video` group. Try finding out which user is running the init script and add it to the `video` group.

closed 2022-10-14 by jacksonliam

SaintTaint, 2022-10-18:
Thank you.  I'll try it out and get back to you.