jacksonliam/mjpg-streamer #375: Buffer too small

thomas-pegot, 2022-12-25:
I am using esp32-camera module to send through http to orange pi zero (dietpy)

Command: `mjpg_streamer -i "input_http.so -H 192.168.12 -p 81" -o "output_http.so -p 8082`

Output : 
```
: Success
Buffer too small
: Success
Buffer too small
...```
The code related [here](https://github.com/jacksonliam/mjpg-streamer/blob/310b29f4a94c46652b20c4b7b6e5cf24e532af39/mjpg-streamer-experimental/plugins/input_http/mjpg-proxy.c#L94)

So the code for sending jpg chunk comes from espressif/esp32-camera and I took a look here : https://github.com/thomas-pegot/esp32-camera/blob/77f763038e76160f0a78a0f484d4d9788f9d915b/conversions/to_jpg.cpp#L212

Apparently the buffer is `128*1024` while the max buffer in mjpg-streamer is `BUFFER_SIZE 1024 * 256` so it should be good enough.

jacksonliam, 2022-12-26:
This input plugin is only really made for talking to other instances of mjpg streamer. The esp would have to send the data including the content length header exactly as mjpg streamer does.

closed 2022-12-26 by jacksonliam

thomas-pegot, 2022-12-26:
The content sent is this : https://github.com/thomas-pegot/camera_web_server/blob/a30dacecc9ce6440874142d6b8ab1d6d26ffe6b4/src/app_httpd.c#L64

I can change variable such that it's exactly what mjpg-streamer is expecting.