jacksonliam/mjpg-streamer #379: Can't make it work on ubuntu server

adetogni, 2023-01-17:
Hi, I have ubuntu server running on a VM on a qnap nas. A Trust Tyro webcam is connected in USB and lsusb command can see it

Bus 001 Device 006: ID 145f:02aa Trust Trust Full HD Camera

however, when I try to launch mjpg-streamer I get "bind address already in use"

The ip of the NAS is 192.168.1.201 but the IP of the VM is 192.168.1.155. 
On 192.168.1.155:5000 there's octoprint server running and it works perfectly

If I go to 192.168.1.155:8080 I get a 
`
501: Not Implemented!
no www-folder configured

`

(this is by the way strange. Until yesterday, I was getting the mjpg-streamer test page and now I have an error...)

haproxy has this configuration:

`frontend public
    bind *:80
    use_backend webcam if {path_beg /webcam/ }
    default_backend octoprint

backend webcam
    http-request replace-path /webcam/(.*)
     server webcam1 127.0.0.1:8080`

I'm pretty sure that I'm doing something wrong with the ports...can someone help me?

jacksonliam, 2023-01-17:
Something is using the port when you try to launch mjpg streamer.

However this is not the right place for individual support questions, please ask on a forum or Reddit or something.

closed 2023-01-17 by jacksonliam

CodeWithCJ, 2024-05-10:
Did you figure it out? I am also facing similar issue as the port is used by another application. Any idea how to change the port?

adetogni, 2024-05-10:
Hi @CodeWithCJ in the end I decided to simply buy a logitech webcam and it was working. But ALSO I bought a Raspi and made it work that way. The VM on the NAS was problematic since the webcam had to constantly being reconnected on every restart of the VM plus a series of minor inconveniences on performances. So I decided to simply buy a raspi.