jacksonliam/mjpg-streamer #380: Does not work with octoprint

adetogni, 2023-01-19:
Seems that mjpg-streamer is confliting in some way with octoprint

1) On Ubuntu server 22.04 I installed mjpg-streamer and tested it. Everything was working fine. So I stopped mjpg-streamer
2) Then I installed octoprint which connects to a device on ttyUSB0 (the 3d printer)
3) Then once octoprint service is running I launched again mjpg-streamer. The test page works and I can see a screenshot but then it stops working and in console I receive a lot of errors:

[260.4302408] xhci_hcd 0000:00:04.0: ERROR Transfer event TRB DMA ptr not part of current TD ep_index 2 comp_code 13

this line appears multiple times

hitting Ctrl-C won't stop, it stays on "deallocating resources" forever. 

This is a VM running on a QNAP server. Until I installed octoprint, it was working fine. I can provide the VM and the snapshot.

If I disconnect the 3d printer usb cable (disconnect from VM administration, not phisically disconnect) the error still appears

lsusb reports the camera as
BUS 001 Device 002: ID 145f:02aa Trust Full HD Webcam

adetogni, 2023-01-19:
no, I stand corrected. I reverted the status of the VM to when octoprint was not installed and the webcam is NOT working. I can make it work for a few minutes and then I still receive the same errors.

adetogni, 2023-01-20:
the webcam is a Trust Tyro. I am now going to test with a different webcam, hopefully soon

jacksonliam, 2023-01-20:
Unlikely to do with mjpg-streamer, must be hardware or camera driver issue.

closed 2023-01-20 by jacksonliam

adetogni, 2023-01-23:
I can confirm, the Trust Tyro was the issue. I switched to a Logitech C920 and everything went fine. cheers