jacksonliam/mjpg-streamer #384: Stream-Resolution: 640x480 works, but snapshot has: 4056x3040 (hq-cam/picam3 via libcamera)

Dephrilibrium, 2023-03-05:
Hi there,

firstly, sorry when I'm wrong, because tbh I'm using the [arducam-fork](https://github.com/ArduCAM/mjpg-streamer) but there issues are deactivated, so I hoped I can find help here.

I adjusted my libcamera-stream (via mjpg-streamer) to an resolution of 640x480, which works as expected. As I tried to use the built-in timelapse (not octolapse!) I noticed that the snapshot resolution is the full camera-sensor resolution (4056x3040) instead of 640x480 (which I had expected). I'm not exactly sure why, but that causes some problems with missing frames and ffmpeg crashes. I suggest, that the snapshot tries to start/open another camera-stream, because 640x480 uses 
```
[2:56:05.610211948] [12386]  INFO Camera camera.cpp:1028 configuring streams: (0) 680x480-BGR888
[2:56:05.611685437] [12383]  INFO RPI raspberrypi.cpp:851 Sensor: /base/soc/i2c0mux/i2c@1/imx477@1a - Selected sensor format: 2028x1080-SBGGR12_1X12 - Selected unicam format: 2028x1080-pBCC
```
So the stream can't provide 4056x3040, due to the sensor is 2x2 binned to the "half" resolution.

I also had a look on the configuration of OctoPrint, but there was no configuration-line for snapshot-parameters.

Specs:
Hardware: Raspberry Pi 3
OS: Raspbian-OS Lite (based on debian 11 - bullseye)
Connection: WLAN
Software:
- Python3 (v3.9.2)
- OctoPrint (v1.8.7)
- [mjpeg-streamer (arducam-fork)](https://github.com/ArduCAM/mjpg-streamer), followed the [setup in the arducam-forum](https://forum.arducam.com/t/using-libcamera-imx519-on-octopi/2272/2#download-libcamera-plugin-source-code-7)
- libcamera for HQ-Camera (picamera3) controls

So, I'm not exactly sure, about how to configure the resolution (or even better - just taking a frame from the already running stream) for a snapshot.

Kind regards,
Deph

jacksonliam, 2023-03-05:
This fork doesn't have a libcamera plugin and I have no idea how the fork's one works. But I believe it uses software jpeg encoding so the frames it is providing could be any size independent of the camera resolution. If it is not obeying you're requested resolution you will have to raise that with them.

The snapshot URL provides the last jpeg frame that would normally be output in a stream, it's impossible for it to be a different resolution to the stream. 

I haven't checked, but there might be some memory corruption or something to the snapshot frame if you're using high framerates with large frame sizes.

closed 2023-03-05 by jacksonliam

Dephrilibrium, 2023-03-10:
Hi jacksonliam,

firstly, sorry for the late response.

> This fork doesn't have a libcamera plugin and I have no idea how the fork's one works. But I believe it uses software jpeg encoding so the frames it is providing could be any size independent of the [...]
you're using high framerates with large frame sizes.

Sure it has. Its sources can be found in:
[mjpg-streamer/mjpg-streamer-experimental/plugins/input_libcamera](https://github.com/ArduCAM/mjpg-streamer/tree/master/mjpg-streamer-experimental/plugins/input_libcamera)
When you compile the repository a `input_libcamera.so` file is created in `/home/haum/mjpg-streamer-libcamera/mjpg-streamer-experimental` path which can be used for the HQ-Cam.

But anyway. I hope there will be some updates from arducams site in future.
Btw. is it planned that you will add the HQ-Cam also into your project?

Kind regards,
Deph