jacksonliam/mjpg-streamer #386: Bullseye can't install `libjpeg8-dev` - substituting `libjpeg62-turbo-dev` succeeds

Paraphraser, 2023-03-16:
Just for information, trying to install the `libjpeg8-dev` prerequisite on a 4GB Raspberry Pi 4 Model B Rev 1.1 running Debian GNU/Linux 11 (bullseye) full 64-bit OS fails:

```
$ uname -a
Linux tri-dev 5.15.84-v8+ #1613 SMP PREEMPT Thu Jan 5 12:03:08 GMT 2023 aarch64 GNU/Linux

$ sudo apt install libjpeg8-dev      
Reading package lists... Done
Building dependency tree... Done
Reading state information... Done
Package libjpeg8-dev is not available, but is referred to by another package.
This may mean that the package is missing, has been obsoleted, or
is only available from another source
However the following packages replace it:
  libjpeg62-turbo-dev:armhf libjpeg62-turbo-dev

E: Package 'libjpeg8-dev' has no installation candidate
```

Accepting the hint, installing `libjpeg62-turbo-dev` and following the remaining README instructions builds `mjpg_streamer` successfully.

The camera I'm working with (Logitech, Inc. HD Pro Webcam C920) then streams OK.

This may be related to #296 and #190.

Bazmundi, 2023-05-27:
Yep, so I am running Jessie 8 on an old Atom-based Acer Aspire One, so having to run Jesse 8.  So, looks like there are all sorts of problems with dependencies, so there also seems then no distinct set of workarounds currently for Jesse 8.  I will try to document best I can, if I can, the path to install.  But it isn't as simple as libjpeg62-turbo-dev.

pm-pm, 2023-06-19:
Successfull here on fresh released debian 12 "bookworm". :-)
Thanks.