jacksonliam/mjpg-streamer #395: "input_raspicam" is not compiled

Dephrilibrium, 2023-09-17:
Heyho,

i tried to install the latest octoprint with "native" HQ-Cam support. For that, the installation [description for manual installation](https://community.octoprint.org/t/setting-up-octoprint-on-a-raspberry-pi-running-raspberry-pi-os-debian/2337) links to your experimental mjpg-streamer repo.

When I run the build via make, my output skips a bunch of plugins and only builds the plugins:
- input_file.so
- input_http.so
- input_uvc.so
- output_file.so
- output_http.so
- output_rtsp.so
- output_udp.so

Within the CMakeLists.txt, the raspi-plugin is added to the build-targets:
```
# Input plugins
add_subdirectory(plugins/input_file)
add_subdirectory(plugins/input_http)
add_subdirectory(plugins/input_opencv)
add_subdirectory(plugins/input_raspicam)    <---
add_subdirectory(plugins/input_ptp2)
add_subdirectory(plugins/input_uvc)
```

Unfortunately I'm not familiar with cmake, That's why I'm not  sure about why some plugins are not built. May you have an idea to fix this.

Thanks in advance!

Best regards,
Deph

Dephrilibrium, 2023-09-17:
Heyho again,

just solved the problem as I did a bit more search via google. By this I found Issue https://github.com/jacksonliam/mjpg-streamer/issues/259 of this repo.

This comment https://github.com/jacksonliam/mjpg-streamer/issues/259#issuecomment-1024678626 explains what to do

The point with Raspbian64 and the raspicam-plugin is, that they moved a few dependencies from `/opt/vc` into `/usr` for more "standardization" (Ref to: https://forums.raspberrypi.com//viewtopic.php?t=275370 mentioned by https://github.com/jacksonliam/mjpg-streamer/issues/259#issuecomment-655197554). So, go into the raspicam-plugin folder and adjust the dependency paths within the CMakeLists.txt file.

Short fix: Assuming, that you have entered the repositories' root in bash you can run:
```
sed -i 's_/opt/vc_/usr_g' mjpg-streamer-experimental/plugins/input_raspicam/CMakeLists.txt
```

Kind regards,
Deph

AkitaAttribute, 2023-09-18:
+ the need to fix this.

Dephrilibrium, 2023-09-22:
Heyho again,

I found a [thread](https://raspberrypi.stackexchange.com/questions/34107/libmmal-core-so-missing/34110#34110) where it's mentioned to use `sudo rpi-update` and check if one line of the terminal-output says something like `video-libraries updated`.

As I searched for the lib-files I think I found them under `/opt/vc/lib/*`:
```
<user>@<host>:~/mjpg-streamer/mjpg-streamer-experimental $ sudo find / | grep mmal_
...
/opt/vc/lib/libmmal.so
/opt/vc/lib/libmmal_vc_client.so
/opt/vc/lib/libmmal_components.so
/opt/vc/lib/libmmal_util.so
/opt/vc/lib/libmmal_core.so
...
```

So the libs seem to be now available under the original paths (**not** replacing any paths in the `CMakeLists.txt` as described in my [earlier post](https://github.com/jacksonliam/mjpg-streamer/issues/395#issuecomment-1722439530)).

Then I tried a full rebuild under using:
```
make clean
make distclean
make
```
but now the output returns, that the libraries are incompatible and then they aren't found again:
```
[ 44%] Linking C shared library input_raspicam.so
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_core.so when searching for -lmmal_core
/usr/bin/ld: cannot find -lmmal_core
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_core.so when searching for -lmmal_core
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_util.so when searching for -lmmal_util
/usr/bin/ld: cannot find -lmmal_util
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_util.so when searching for -lmmal_util
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_vc_client.so when searching for -lmmal_vc_client
/usr/bin/ld: cannot find -lmmal_vc_client
/usr/bin/ld: skipping incompatible /opt/vc/lib/libmmal_vc_client.so when searching for -lmmal_vc_client
/usr/bin/ld: skipping incompatible /opt/vc/lib/libvcos.so when searching for -lvcos
/usr/bin/ld: skipping incompatible /opt/vc/lib/libbcm_host.so when searching for -lbcm_host
collect2: error: ld returned 1 exit status
```

Maybe someone has an idea of what's going on.

Kind regards,
Deph

xia2206636330, 2024-01-25:
Has this problem been resolved? How to solve it, mine is the latest version of Raspberry Pi 32-bit

hellbe, 2024-01-30:
Same problem here, getting the same as @Dephrilibrium's latest post. The files are found but stated to be incompatible as it says above.

spectroman, 2024-02-06:
for me the suggestion:

```sed -i 's_/opt/vc_/usr_g' mjpg-streamer-experimental/plugins/input_raspicam/CMakeLists.txt```

made the compiler generate the plugin properly and it ran with no problems as well...

it gave some warning on compiling time, but no fatalities:

[image]

and later:

`mjpg_streamer -i "input_raspicam.so -x 960 -y 720" -o "output_http.so -w ./www"`

ran as expected.

Resousse, 2024-02-15:
On my RPI, freshly upgraded to bookworm, there is no libmmal.so at all...
I went from buster to bullseye and directly from bullseye to bookworm