jacksonliam/mjpg-streamer #397: Raspberry Pi arm64 potential fix

MattiKemp, 2023-12-14:
For those having trouble getting their camera to work with newer Raspberry Pis (works on Ubuntu 20.04), try this:

install mjpg-streamer dependencies

`sudo apt-get update`
`sudo apt-get install cmake libjpeg8-dev`
`sudo apt-get update`
<br>
install aarch64 gcc and g++ for userland

`sudo apt-get install gcc-aarch64-linux-gnu g++-aarch64-linux-gnu`
`sudo apt-get update`
 <br>

Thanks to [https://github.com/waveform80/picamera/issues/540#issuecomment-881645216](https://github.com/waveform80/picamera/issues/540#issuecomment-881645216):

download userland

`git clone https://github.com/raspberrypi/userland`
`cd userland`
`git revert f97b1af1b3e653f9da2c1a3643479bfd469e3b74`

press ctrl+o, enter, crtl+x, and enter

`git revert e31da99739927e87707b2e1bc978e75653706b9c`

press ctrl+o, enter, crtl+x, and enter
<br>

Fix /interface/mmal/CMakeLists.txt

`sed '/^target_link_libraries(mmal mmal_core mmal_util mmal_vc_client vcos mmal_components)*/i SET(CMAKE_SHARED_LINKER_FLAGS "-Wl,--no-as-needed")' ./interface/mmal/CMakeLists.txt > ./interface/mmal/CMakeLists-temp.txt`
`mv ./interface/mmal/CMakeLists-temp.txt ./interface/mmal/CMakeLists.txt`
<br>

Build userland

`sudo ./buildme --aarch64`
<br>

Copy userland libraries

`sudo cp /opt/vc/lib/*.so /usr/lib/aarch64-linux-gnu/`
`cd ..`
<br>

Build and install mjpg-streamer

`git clone https://github.com/jacksonliam/mjpg-streamer.git`
`cd mjpg-streamer`
`cd mjpg-streamer-experimental`
`make distclean`
`make CMAKE_BUILD_TYPE=Debug`
`sudo make install`
<br>

Check if your camera is detect and enabled

`vcgencmd get_camera`

This should return: supported=1 detected=1
If it doesn't:

`sudo vim /boot/firmware/config.txt`

and add start_x=1 to the very end of the file, save, then reboot. 
<br>

If you eventually get: supported=1 detected=1, return to the /mjpg-streamer/mjpg-streamer-experimental/ directory, and try:

`sudo chmod +x ./start.sh`
`./start.sh`

If that doesn't work, while you are still in /mjpg-streamer/mjpg-streamer-experimental/:

`export LD_LIBRARY_PATH="$(pwd)"`
`./mjpg_streamer -i "input_uvc.so -y -r 1280x720 -f 20" -o "output_http.so -w `pwd`./www"`
<br>

This is the only thing that got it working for me on a Raspberry Pi 4b running arm64 Ubuntu (20.04). I hope something in this mess may help you.

Resousse, 2024-02-15:
Thank you very much! It works.
All the userland libraries disappeared before...

Debcharon, 2024-04-14:
[image]

```plain-text
# /boot/firmware/config.txt

# Automatically load overlays for detected cameras
#camera_auto_detect=1
camera_auto_detect=0
...
# Don't have the firmware create an initial video= setting in cmdline.txt.
# Use the kernel's default instead.
#disable_fw_kms_setup=1
...
[all]
start_x=1
```

```bash
root@raspberrypi:~# cat /etc/*release
PRETTY_NAME="Debian GNU/Linux 12 (bookworm)"
NAME="Debian GNU/Linux"
VERSION_ID="12"
VERSION="12 (bookworm)"
VERSION_CODENAME=bookworm
ID=debian
HOME_URL="https://www.debian.org/"
SUPPORT_URL="https://www.debian.org/support"
BUG_REPORT_URL="https://bugs.debian.org/"
root@raspberrypi:~# uname -a
Linux raspberrypi 6.6.20+rpt-rpi-v8 #1 SMP PREEMPT Debian 1:6.6.20-1+rpt1 (2024-03-07) aarch64 GNU/Linux
```

It really help me, thanks for your contributing, I'm confused for this problem been 2 days.

Here I'll attach some output from terminal.

[build.txt](https://github.com/jacksonliam/mjpg-streamer/files/14969604/build.txt)