jacksonliam/mjpg-streamer #405: streaming multiple usb cameras possible on pi 4B?

cupofnestor, 2024-08-08:
I am using octoprint-docker to run multiple instances of octoprint on a single RaspBerry PI.  I've set up a bash script to run mjpg_streamer on the host with a number of webcams attached to the PI.

```bash
#!/usr/bin/env bash
source /home/ubuntu/streamer/stream_env
I=1
RES="640x480"
FRATE="30"
PORT="8080"
ADD_FLAGS="-y "
OUT_ARG="$OUTPUT_HTTP -l 0.0.0.0 -p ${PORT}"
/snap/bin/mjpg-streamer -i "$INPUT_UVC ${ADD_FLAGS}  -r ${RES} -f ${FRATE}  -d ${VIDEO_DEVICES[0]}"  -i "$INPUT_UVC ${ADD_FLAGS}  -r ${RES} -f ${FRATE} -d ${VIDEO_DEVICES[1]}"  -o "$OUT_ARG" -b
```
where $VIDEO_DEVICES is an array of dev paths e.g.  /dev/video0 , although in this case I am using `/dev/video/by-path` paths.

When I run the script without the `-b` flag, this is the output:

```
MJPG Streamer Version: git rev: f387bb44e6c087271b763b27da998bf2e06c4f5d
 i: Using V4L2 device.: /dev/video1
 i: Desired Resolution: 320 x 240
 i: Frames Per Second.: 30
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 i: Using V4L2 device.: /dev/video0
 i: Desired Resolution: 320 x 240
 i: Frames Per Second.: 30
 i: Format............: YUYV
 i: JPEG Quality......: 80
 i: TV-Norm...........: DEFAULT
UVCIOC_CTRL_ADD - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_ADD - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt (relative): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Pan/tilt Reset: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Focus (absolute): Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Mode: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at LED1 Frequency: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Disable video processing: Inappropriate ioctl for device (25)
UVCIOC_CTRL_MAP - Error at Raw bits per pixel: Inappropriate ioctl for device (25)
 o: www-folder-path......: disabled
 o: HTTP TCP port........: 8080
 o: HTTP Listen Address..: 0.0.0.0
 o: username:password....: disabled
 o: commands.............: enabled
```

The second stream (  http://my.domain/?action=stream_1 ) either takes forever to load, or else never does.  If I use `320x240` resolution, I can see both streams but they flicker every few seconds. 

`htop` shows two mjpg_streamer instances each using around 8% of the CPU, and memory looks good overall.

Is there any reason that this hardware should not be able to handle streaming 2 or more webcams?

username7291, 2024-08-09:
Start one mjpg streamer with port 8080
Start another with another port. 
" i: Format............: YUYV"  better use MJPEG it doesnt need to be converted, if your cams not very old. Make sure its supported by Linux, some cams dont work or deliver broken images.
If anything works you can try again with just 1 mjpg-streamer.
For the Hardware 4 cams with 640x480(mjpeg) is not a problem i would guess 1280x720 works too.

jacksonliam, 2024-08-09:
I've had two cameras working with much worse hardware so this is not a reproducible issue.

I assume one camera at a time is working fine? What about recording from both at once with different software? As it could be camera driver related. 

If they don't work simultaneously with other software it could be a power issue, maybe try a powered usb hub?

closed 2024-08-09 by jacksonliam